IPNL-POLYU/UrbanNavDataset #4: RINEX navigation files

kcwian, 2021-11-03:
Hi, 
I have a question regarding RINEX files that you provide with the dataset recorded in Hong Kong. There are two types of files available: **_.obs_** and **_.nmea_**, but if I understand correctly, libraries such as RTKLIB require additionaly **_.nav_** or **_.n/.b_** navigation files, that contain ephemeris data, which is necessary to calculate satellites position. Is it possible to extract such data from **_.nmea_** files or is there any other way to obtain that data?

DarrenWong, 2021-11-03:
You can get the broadcast ephemeris data from https://www.geodetic.gov.hk/en/rinex/downv.aspx

A tutorial is also provided here:[Getting Started](https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/docs/GETTING_STARTED.md)

na851998, 2022-07-19:
Hi,
Can you share how the nav files in tokyo dataset are obtained?

DarrenWong, 2022-07-19:
You can donwload the zipped files and get the nav files in tokyo dataset.

[image]

na851998, 2022-07-20:
Sorry, I should be more specific.
Currently, I'm trying to provide GINav - an open-source software on Github raw data from Hong Kong dataset. This software already used the Tokyo dataset and it worked well. However, it is not the case with Hong Kong dataset. 
I followed the tutorial to obtain the nav file of Hong Kong dataset and noticed that the size of the nav file was much smaller compared with nav file in Tokyo dataset. So I was wondering how the Tokyo's nav file is obtained so I can do the same processing with Hong Kong's nav file

DarrenWong, 2022-07-21:
Can you provide the Hong Kong Nav files which you are using? The size might be different as the Toyko dataset provide the whole day mixed navigation files and you might download only one hrs data from the Hong Kong Lands Department.

na851998, 2022-07-21:
This is the nav file I am using, when I process it with the obs file using RTKLib, the result was quite reasonable.
[New Text Document.txt](https://github.com/IPNL-POLYU/UrbanNavDataset/files/9156207/New.Text.Document.txt)
I tried to download the whole day nav file from Hong Kong Lands Department, its size was still really small compared with nav file from Tokyo dataset. 
So I have a question, was the Tokyo's nav file obtained by downloading from the Internet or something else?

DarrenWong, 2022-07-21:
About the size, you can see your files are 366 rows with GPS only (1 hr internal)
The Tokyo data is a mixed navigation file that includes GPS/GLO/GAL/BDS/QZS/SBAS/IRNSS. You can delete the satellite system you are not using to reduce the size.

For the question of the Tokyo's nav file obtained by downloading from the Internet or something else? I need to invite our colleagues from Japan to answer your question. We will update later.

na851998, 2022-07-21:
I just found out that the problem was not the nav file, it was the configuration file of the software.
Thank you for your time answering my questions.

taroz, 2022-07-22:
Sorry for the late reply. It seems that the problem has already been solved, but I would like to explain about the navigation file for Tokyo dataset.

The navigation file for Tokyo dataset was downloaded from the IGS website, and it is large because it contains navigation data for all satellites that are not included in the observation file. The navigation files can be downloaded from the following links.
https://igs.org/mgex/data-products/#bce

DarrenWong, 2022-07-22:
@na851998  great to hear that you solve your problem. Thanks Taro for the explanation.

closed 2022-07-22 by DarrenWong