IPNL-POLYU/UrbanNavDataset #6: are all the sensors synchronized to the same time reference

zouyajing, 2022-01-04:
Thanks a lot for your great works.
I have two questions:
a. Are all the sensors synchronized to the same time reference using the PPS signal? Do we need to estimate the time delay in the algorithm?
b. I want to buy some cameras for urban positioning. It will be much appreciated if you could provide the detail about your stereo camera (zed) and mono camera (point grey + lens). I only know the band, but do not know the detail.

Best regards,
YajingZou

DarrenWong, 2022-01-04:
Hey Zou,

Thanks for your interest in our work.

We use Chrony + GPSd to receive the NMEA+PPS signal from Ublox M8T to synchronize ROS computer with GPS time. But the wire latencies between sensors (LiDAR, Camera and IMU) and ROS computer are not considered in this dataset.

For the cameras, we use ZED2 and point grey (BFLY-U3-23S6C-C) with len (f=6mm/F1.8 kowa). You can also try some new models such as ZED2i with IP66 Enclosure

zouyajing, 2022-01-04:
> 
Hey Wong,

Thanks a lot for your quick reply. I will try to buy these cameras.
**But the wire latencies between sensors (LiDAR, Camera and IMU) and ROS computer are not considered in this dataset.**
So PPS signal is not used to trigger the camera or lidar. These sensors are synchronized on software instead of on hardware. Is that right?

Thanks and best regards,
YajingZou

DarrenWong, 2022-01-05:
@zouyajing we use the ROS time to software sync the timestamp of the multi-sensor. For more specific,  the camera and IMU are stamped once arrive. The LiDAR is stamped on the first point in the cloud. 
If you want to develop a hardware synced kit, you can refer to one [research work](http://apolloscape.auto/inpainting.html) from apolloauto with a HESAI [pandora](https://www.datrontechnology.co.uk/wp-content/uploads/2020/03/20190606-Pandora-Brochure.pdf)

zouyajing, 2022-01-05:
> @zouyajing we use the ROS time to software sync the timestamp of the multi-sensor. For more specific, the camera and IMU are stamped once arrive. The LiDAR is stamped on the first point in the cloud. We still working on the PPS siginal synced with multi-sensors. If you want to develop a hardware synced kit, you can refer to one [research work](http://apolloscape.auto/inpainting.html) from apolloauto with a HESAI [pandora](https://www.datrontechnology.co.uk/wp-content/uploads/2020/03/20190606-Pandora-Brochure.pdf)

Hi, Wong. Thanks for your reply. I have no further questions.

closed 2022-01-06 by DarrenWong