IPNL-POLYU/UrbanNavDataset #11: GNSS Blackout in UrbanNav-HK-Medium-Urban-1

tipf, 2022-03-03:
Hi everyone,
there seems to be a GNSS blackout in the "20210517.light-urban.tste.ublox.f9p.obs" file of the "medium urban" dataset.
Exactly between:
`> 2021  5 17  2 42 30.0050000  0 28  `
and
`> 2021  5 17  2 44 41.0050000  0 29   `
During this period, no observation exists in the RINEX file. 
Other files like "20210517.light-urban.tste.ublox.f9p.splitter.obs" or the m8t files do have observations in this period.

Is this a known property of the dataset or a problem in the post-proccessing?

Best Regard
Tim

DarrenWong, 2022-03-04:
Dear Tim, 

Yes, some receivers cannot receive signals well in short periods in urban canyons. You can try other sensors if you want to have continuous measurements, e.g, _20210517.light-urban.tste.ublox.f9p.splitter.obs_ which another F9P uses a geodetic antenna 

Best Regards,
Darren

tipf, 2022-03-04:
Dear Darren,
thanks for your fast response. I plotted the visibility of all satellites over time and there is a big gap of 120s.
I understand that such things can happen in urban canyons, but 120s while driving in a medium-urban environment?
Isn't it unlikely that all satellites drop out simultaneously?

I added furthermore a plot of the pseudorange residual over time. You can see the phenomenon there too, and there is no increase in the error before and after the blackout.
Could it be a receiver malfunction or a glitch in the data recording?

[image]

[image]

DarrenWong, 2022-03-04:
Dear Tim,

Yes, I think this F9P sensor very possibly has some receiver issues in these 120s. Let me involve one colleague which an expert in GNSS to help this issue @ivannhf 

Best Regards,
Darren

ivannhf, 2022-03-10:
Dear Tim,

Seems there is a malfunction on the receiver, could be a connection lost during that period.
We suggest avoiding that period or using another F9P receiver connect to a geodetic-grade antenna. 

Best regards,
Ivan

tipf, 2022-03-10:
Dear Darren and Ivan,

could you maybe add a note to inform potential users about that glitch?
(And can decide whether they want to avoid it or not.)

Best Regards
Tim

DarrenWong, 2022-03-10:
Dear Tim,

Thank you for your suggestion. Notification of this issue is added to the description of the medium urban. e6c236adf87b48ae1797b2e142049928569c0844

Best Regards,
Darren

tipf, 2022-03-10:
Thanks a lot Darren!