IPNL-POLYU/UrbanNavDataset #12: Missing GT in UrbanNav-HK_Mongkok-20210518_sensors.bag

tipf, 2022-03-03:
Hi again,
it seems like the NovAtel ground truth is missing for the "harsh urban" dataset:

```
rosbag info UrbanNav-HK_Mongkok-20210518_sensors.bag 
path:        UrbanNav-HK_Mongkok-20210518_sensors.bag
version:     2.0
duration:    56:07s (3367s)
start:       May 18 2021 05:06:57.37 (1621307217.37)
end:         May 18 2021 06:03:04.99 (1621310584.99)
size:        142.7 GB
messages:    1652880
compression: none [113896/113896 chunks]
types:       sensor_msgs/CameraInfo    [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image         [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu           [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/PointCloud2   [1158d486dd51d683ce2f1be655c3c181]
             sensor_msgs/TimeReference [fded64a0265108ba86c3d38fb11c0c16]
topics:      /imu/data                        1347064 msgs    : sensor_msgs/Imu          
             /left/lslidar_point_cloud          33681 msgs    : sensor_msgs/PointCloud2  
             /right/velodyne_points             33389 msgs    : sensor_msgs/PointCloud2  
             /time_reference                     3367 msgs    : sensor_msgs/TimeReference
             /velodyne_points                   33647 msgs    : sensor_msgs/PointCloud2  
             /zed2/camera/left/camera_info      50433 msgs    : sensor_msgs/CameraInfo   
             /zed2/camera/left/image_raw        50433 msgs    : sensor_msgs/Image        
             /zed2/camera/right/camera_info     50433 msgs    : sensor_msgs/CameraInfo   
             /zed2/camera/right/image_raw       50433 msgs    : sensor_msgs/Image
```

DarrenWong, 2022-03-04:
Hey Tim,

Thank you for your interest in our work. 

The NovAtel ground truth cannot performs well in harsh urban. Only parts of the GT can be verified (from UTC 1621307670 to 162130998) such that we only provide the GT in a separate file. You can check out the GT files directly from [here](https://www.dropbox.com/s/yaexvwppb6k3szp/UrbanNav_mongkok_GT_part_raw.txt?dl=0) 

[image]
[image]

tipf, 2022-03-04:
Thanks again Darren!
Would it be an option to provide a shorter rosbag that is complete?

Using the datasets gets a bit complicated if the data format is not consistent across the different tracks.
At least a warning before downloading ~150GB without the GT would be nice. :-)

Best Regards
Tim

DarrenWong, 2022-03-04:
1. Would it be an option to provide a shorter rosbag that is complete?

Yes, we will provide a separate link for this by this weekend.

2. Using the datasets gets a bit complicated if the data format is not consistent across the different tracks.
At least a warning before downloading ~150GB without the GT would be nice. :-)

Thank you for your suggestion. We will update the readme with (1) together.

DarrenWong, 2022-03-06:
Dear Tim,

A shorter rosbag is provided [here](https://www.dropbox.com/s/nsvi0cvwkcds0s6/UrbanNav-HK_Mongkok-20210518_sensors_part.bag?dl=0). A [zipped version](https://www.dropbox.com/s/mtkoo1p3ykprnt8/UrbanNav-HK_Mongkok-20210518_sensors_part.zip?dl=0 ) (58GB) is also attached for your reference. Note that the starting point is non-static.

[image]

We also update the readme for the related issue https://github.com/IPNL-POLYU/UrbanNavDataset#urbannav-hk-harsh-urban-1
[image]

Thanks again for your suggestion.

Best Regards
Darren

tipf, 2022-03-09:
Thanks a lot for the new files Darren!

BestRegards
Tim

closed 2022-03-09 by tipf

yinhaosen, 2023-04-12:
The shorter bag link does not work now, dropbox prompts the file was deleted.

DarrenWong, 2023-04-12:
Hey @yinhaosen  thanks for your kind reminder. The link is updated in the readme!

https://github.com/IPNL-POLYU/UrbanNavDataset/commit/275efd1a7da63b920825cd6cc2ea52d0b55f6f6f

ZhenZ163, 2024-12-16:
Hi Darren, could you provide the link of zipped version of shorter rosbag，thanks!

ZhenZ163, 2024-12-16:
Hi Darren, could you provide the Baidu Cloud link of zipped version of shorter rosbag，thanks!

DarrenWong, 2024-12-19:
Thanks for your interest in our dataset, please check the below link for the shorter rosbag in Baidu Cloud Link,

通过网盘分享的文件：urbannav_Mongkok_part
链接: https://pan.baidu.com/s/1h7Vli5w7eI0l9-99wX9ZuA 提取码: wgax 
--来自百度网盘超级会员v3的分享

ZhenZ163, 2024-12-23:
> Thanks for your interest in our dataset, please check the below link for the shorter rosbag in Baidu Cloud Link,
> 
> 通过网盘分享的文件：urbannav_Mongkok_part 链接: https://pan.baidu.com/s/1h7Vli5w7eI0l9-99wX9ZuA 提取码: wgax

Thank you for your help！

ZhenZ163, 2024-12-23:
> Thanks for your interest in our dataset, please check the below link for the shorter rosbag in Baidu Cloud Link,
> 
> 通过网盘分享的文件：urbannav_Mongkok_part 链接: https://pan.baidu.com/s/1h7Vli5w7eI0l9-99wX9ZuA 提取码: wgax

Thank you for your help Darren！