IPNL-POLYU/UrbanNavDataset #14: Span + IE + Rosbag

paaraujo, 2022-03-11:
How did you include the processed reference in the rosbag?

DarrenWong, 2022-03-11:
You can use the ROS API to do so. http://wiki.ros.org/rosbag/Code%20API

paaraujo, 2022-03-11:
Hi @DarrenWong, that is interesting. Doesn't it mess with the timestamps? Do you record the bag during the experiment and add the reference later?

DarrenWong, 2022-03-12:
Yes, we post-processing the ground truth with IE then add it into rosbag. As the reference point has a UTC timestamp, you can assign the timestamp to the message with some code like this,

`bag.write("/novatel_data/inspvax", gt_utc, latest_gt);`

paaraujo, 2022-03-18:
Do you record the reference data using the NovAtel software suite?

DarrenWong, 2022-03-18:
Yes, we need to post-processing the data. You can also use the [Novatel ROS driver](https://github.com/ros-drivers/novatel_span_driver) directly if the real-time output is needed.

paaraujo, 2022-03-18:
Thanks @DarrenWong!

closed 2022-03-18 by paaraujo