IPNL-POLYU/UrbanNavDataset #15: Can't download UrbanNav-HK-Harsh-Urban-1 ROS file

kcwian, 2022-04-19:
Hi, I have a problem with downloading **UrbanNav-HK_Mongkok-20210518_sensors.bag** (147GB) sequence from Dropbox. I've been trying to download it multiple times, but it always suddenly stops with error.
I tried downloading it using both web browser and terminal, but neither method works. The download stops randomly (few times it stopped at around 130 GB). I attach a terminal screenshot from today's attempt, when it finished at 11.70GB.

Could you please verify if the link is working properly? 
Thanks in advance.

[image]
.

DarrenWong, 2022-04-20:
Thanks for interest in our work. The possible reason is that might have a network interrupted somehow within downloading (easily occurs with large files). The link was tested using the Chrome/Dropbox windows app. 

If you still occur a problem, can you try an alternative link to the zipped file (around 80GB) to download it?  Suggest use web browser or dropbox app.
 
 https://www.dropbox.com/s/o9df93suiv3nfx8/UrbanNav-HK_Mongkok-20210518_sensors.zip?dl=0

kcwian, 2022-04-21:
Thank you, I downloaded the .zip file without problems.