IPNL-POLYU/UrbanNavDataset #18: About ground truth

zhanglei1997, 2022-05-31:
About ground truth
Hello, I also noticed another paper of yours:
Point-wise or Feature-wise? Benchmark Comparison of Public Available LiDAR Odometry Algorithms in Urban Canyons.
For the experimental part of this paper, you used two sequences that were previously made in Hong Kong. I would like to know how you used the EVO tool to get the ground truth from the bag and compare it with all the other algorithm results.
Looking forward to your reply.
[image]

zhanglei1997, 2022-05-31:
Thank you very much for your reply! ! !
Regarding the timestamp, I noticed that the frequency of the GT is 1HZ. But the odometer frequency is generally higher than 1hz. In addition, the two cannot be perfectly aligned without hardware time synchronization. So, does EVO automatically solve this kind of problem by interpolation?
Did you output the latitude and longitude separately when running the bag or output the corresponding odometer information directly through conversion?
Looking forward to your reply!

zhanglei1997, 2022-05-31:
When you convert to TUM format, do you need to ensure that the number of poses output by GroundTruth and the algorithm is the same? When using the TUM format, does EVO have to require the same number of poses?

DarrenWong, 2022-06-01:
It's not necessary to be the same number as EVO will find nearest timestamp (https://github.com/MichaelGrupp/evo/blob/051e5bf63195172af58dc8256cc71618f079f224/test/tum_benchmark_tools/evaluate_rpe.py#L278)

zhanglei1997, 2022-06-01:
Thanks for your answer! My understanding of GNSS data and ROS is very shallow and I mainly do offline data processing.
Could you share your results or procedures for exporting GNSS data to Groundtruth in TUM format in the UrbanNav dataset community on github? This result will help me a lot. It can help me compare algorithm performance more conveniently and efficiently.

closed 2022-06-24 by DarrenWong

DarrenWong, 2022-11-23:
@MoonSheepJx all the sequences have GT in rosbag and files in 1HZ. The [files](https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master/tools/evaluation/gt_for_slam) under tools are transformed to local coordinates. But we still need to use t_max_diff=0.1 to evaluate the SLAM performance, as it might large than the default value of 0.01s in EVO.  

We can post-processing the SPAN CPT data to provide a higher frequency  GT (saying 10hz) using Novatel IE software if needed. Which data are you using?

MoonSheepJx, 2022-11-23:
> @MoonSheepJx all the sequences have GT in rosbag and files in 1HZ. The [files](https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master/tools/evaluation/gt_for_slam) under tools are transformed to [...]

Thanks for quick reply, I 've been using `Urban loco 20190117` and `Urban loco 20190316-1`, when i use EVO `t_max_diff` to aligin, all the traj of GT(extracted by topic navsat/odom use` evo_traj bag ... --save_as_tum`) can be algin, but APE and RPE of FAST-LIO2 is very larger than other datasets (M2DGR), but APE and RPE is low when only consider a few frame (no `t_max_diff`). So, I 'm not sure it is caused by timestamp offset, or it is normal.

I'm going to do some more tests next. Thanks for your reply, that may help me much!

DarrenWong, 2022-11-24:
It might have a small timestamp offset (~0.1s) as the GT and sensor are stamped by the data collection computer. We introduce the [hardware sync](https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/docs/GETTING_STARTED.md#time-synchronization-between-gps-time-and-ros-time) + IE post-processing in the UrbanNav data collection, which is more robust in terms of accuracy and time-sync.

But we still observed some GT provided by SPANCPT in the deep urban areas is not accurate in cm level, especially in height, you can check the quality indicator in the text file or the novatel_msgs