IPNL-POLYU/UrbanNavDataset #20: u-blox extrinsic info

paaraujo, 2022-08-24:
Hello @DarrenWong, do you have the extrinsic measurements from the u-blox receivers to the MTi and to the Reference?

DarrenWong, 2022-08-25:
Dear paaraujo, you can get extrinsic from here https://github.com/IPNL-POLYU/UrbanNavDataset/blob/c95e9bce7cbf444a2e95765ac59409c66e7acb09/UrbanNav-HK-Medium-Urban-1/extrinsic.yaml#L60

Be noted that the extrinsic between the antenna and the reference frame is measured manually with an accuracy of within 5 centimeters. You can optimize the translation by comparing the RTK fixed results and the output from the ground truth.

paaraujo, 2022-08-25:
I didn't find anything related to the u-blox sensors, that is why I asked. According to the drawing shown on the front page, there is a lever arm as well. However, it is a good idea to use the output of the RTKLib to estimate that. Thank you.

DarrenWong, 2022-08-25:
We only calibrate one average result of all the antenna of receivers in the front of the vehicle based on the Y-axis and Z-axis. 
And we also provide a measurement from ublox f9p whcih share the same antenna (rear of the vehicle) with Ground truth with the name "* f9p.splitter*", the approximately lever arm (GNSS->Reference) is   x=0.000, y=-0.560, z=-0.070 m (x-right, y-fwd, z-up)

paaraujo, 2022-08-25:
Alright, I didn't know about the splitter. I will test it, thank you!

paaraujo, 2022-09-01:
Hi @DarrenWong , what about the lever arm between the FlexPak6 and the reference?

DarrenWong, 2022-09-02:
The same as the F9P splitter,  x=0.000, y=-0.560, z=-0.070 m (x-right, y-fwd, z-up). We use a GNSS splitter to receive the measurements and then split them to F9P, FlexPak6, and GroundTruth.

DarrenWong, 2022-09-05:
I am closing the issue as it has no further updates. Please feel free to ask if have any questions.

closed 2022-09-05 by DarrenWong