IPNL-POLYU/UrbanNavDataset #21: About the value z of UrbanNav-HK-Medium-Urban-1 dataset

fentuoli, 2022-09-05:
[image]
Hi, I'm sorry to bother you. I found that  as the vehicle moves, the height of the point scanned by the laser gradually becomes negative.
The picture above is a few frames I took after the bag played for a while, the red point in the picture represent the pose of the current laser, and I set z to 0.
I don't understand why the heights of the scanned points are all negative. Maybe this question is naive, but it really confuses me and prevents me from capturing point clouds with a fixed height range.
Hoping to your reply. Thanks a lot!

fentuoli, 2022-09-05:
Sorry, I found out that it was something wrong with my point cloud transformation function that caused the z of the original points to change, I'm so sorry to bother you!

DarrenWong, 2022-09-05:
Glad to see you solved your problem! the RVIZ indicates that the points have a positive height which scanned the buildings, as shown in the figure below,

[image]

closed 2022-09-05 by DarrenWong

fentuoli, 2022-09-05:
> Glad to see you solved your problem! the RVIZ indicates that the points have a positive height which scanned the buildings, as shown in the figure below,
> 
> [image]

I'll have a check! Thanks a lot!