IPNL-POLYU/UrbanNavDataset #22: The format about novatel_msgs/INSPVAX

fentuoli, 2022-09-11:
Hi, i'm sorry to bother you.When i tried to generate gt of [UrbanNav-HK-Data20200314] from rostopic /novatel_data/inspvax, i found the format of novatel_msgs/INSPVAX is as following:
[image]
but the UrbanNav_TST_GT_raw.txt you provided is as following:
[image]
What does H-Ell mean? The latitude in UrbanNav_TST_GT_raw.txt has three columns, how to get three columns from novatel_msgs/INSPVAX?
If i want to add UTCTime, do I just need to get the current time of /imu/data when writing the novatel_msgs information?

DarrenWong, 2022-09-12:
The H-Ell is short for Ellipsoidal Height from the output of the Novatel. You can approximate it as altitude in /novatel_data/inspvax
The three columns stand for degrees, minutes and seconds of the latitude. If you would like to change the latitude back to D M S, you can refer to the code here https://github.com/IPNL-POLYU/UrbanNavDataset/blob/c95e9bce7cbf444a2e95765ac59409c66e7acb09/tools/gt_vis/convert_novatel_to_pose.py#L53

Yes, you can use IMU timestamp to write Novatel msg information in [UrbanNav-HK-Data20200314]. 
Or you can also convert the GPS time to UTC refer to [here](https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/docs/GETTING_STARTED.md#time-synchronization-between-gps-time-and-ros-time). Noted that the pilot data [20190428] and [20200314] is not fully time synced such that might have a small time offset between GPS Time and UTC.

fentuoli, 2022-09-12:
> The H-Ell is short for Ellipsoidal Height from the output of the Novatel. You can approximate it as altitude in /novatel_data/inspvax The three columns stand for degrees, minutes and seconds of the [...]

Thanks for your quick reply, I will have a try soon!

closed 2022-09-28 by DarrenWong