IPNL-POLYU/UrbanNavDataset #23: 点云地图

ZhaoXiaoyu01, 2022-10-25:
您好，想请教一下如何才可以得到您给出的点云地图真值数据？

DarrenWong, 2022-10-25:
对的，目前点云地图是通过真值拼接的。我们也在尝试采用图优化的方法把真值和LiDAR融合在解决一些场景真值不准的场景。

ZhaoXiaoyu01, 2022-10-25:
谢谢您的解答，想另外问一下我们的真值数据是3轴的还是6轴的，为什么我通过下面代码中的方式结算的姿态是有问题的，是我的方式不对吗？
`    void parse(novatel_msgs::INSPVAXConstPtr msg) {
        Eigen::AngleAxisd rollAngle(msg->roll, Eigen::Vector3d::UnitX());
        Eigen::AngleAxisd pitchAngle(msg->pitch, Eigen::Vector3d::UnitY());
        Eigen::AngleAxisd yawAngle(msg->azimuth, Eigen::Vector3d::UnitZ());
        Eigen::Quaterniond q = yawAngle*pitchAngle*rollAngle;
    }`

DarrenWong, 2022-10-25:
Novatel 给出的真值跟ROS里面定义有点不一致， roll, pitch和azimuth的定义可见:
https://hexagondownloads.blob.core.windows.net/public/Novatel/assets/Documents/Bulletins/apn037/apn037.pdf

你也可以通过python脚本把novatel转成local frame (LiDAR/IMU)
https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master/tools/gt_vis

ZhaoXiaoyu01, 2022-10-25:
十分感谢

closed 2022-10-25 by DarrenWong