IPNL-POLYU/UrbanNavDataset #25: Vinsfusion on UrbanNav-HK-Data20190428

nothing371442, 2022-10-26:
Hi, Thank you for your work very much.
I have a question that the performance of VINS-Fusion on "UrbanNav-HK-Data20190428" is very bad as picture below. I followed the code from https://github.com/baaixw/VINS-Fusion-for-UrbanNavDataset-Evaluation , and change the config as https://github.com/IPNL-POLYU/UrbanNavDataset/issues/2 suggests.
[image]

Would you mind telling me what problem may be?
Thank you very much. Best wishes.

DarrenWong, 2022-10-26:
The VINS Fusion should be fine to run the 0428 data. You might need to revise the config of intrinsic and extrinsic if you are following the latest config from https://github.com/baaixw/VINS-Fusion-for-UrbanNavDataset-Evaluation

nothing371442, 2022-10-26:
> The VINS Fusion should be fine to run the 0428 data. You might need to revise the config of intrinsic and extrinsic if you are following the latest config from https://github.com/baaixw/VINS-Fusion-for-UrbanNavDataset-Evaluation

Thanks. I used the .yaml followed as you provided in https://github.com/IPNL-POLYU/UrbanNavDataset/issues/2, and also changed the intrinsic and extrinsic. But it did not work. 
[cam0_pinhole_lzp.txt](https://github.com/IPNL-POLYU/UrbanNavDataset/files/9868754/cam0_pinhole_lzp.txt)
[vi_car_lzp.txt](https://github.com/IPNL-POLYU/UrbanNavDataset/files/9868755/vi_car_lzp.txt)

DarrenWong, 2022-10-26:
Your config should be fine, as tested in the VINS fusion (figure below) or VINS mono. You can try using the default version of VINS fusion for the old data, as the version might have some revision based on our camera. https://github.com/HKUST-Aerial-Robotics/VINS-Fusion
[image]