IPNL-POLYU/UrbanNavDataset #27: zed2 到 imu 的外参

ZhaoXiaoyu01, 2022-10-28:
您好，从机械结构上看 zed2 与 imu 之间的距离较小，而给出的外参中，z轴平移方向大小为-3.0104266183335913e+00，外参是否准确？
[image]

baaixw, 2022-10-30:
同学你好，请参考这一个issue:
https://github.com/IPNL-POLYU/UrbanNavDataset/issues/17#issuecomment-1131292203

ZhaoXiaoyu01, 2022-10-30:
您好，[#17](https://github.com/IPNL-POLYU/UrbanNavDataset/issues/17#issuecomment-1131292203) 好像并没有回复或处理关于平移部分外参的问题
[image]

baaixw, 2022-10-31:
同学，这个Issue是在讨论外参平移部分的误差，并且很详细的讨论了X轴方向的误差。现在可以帮到你的建议就是：你可以把我们的外参当作初始值，然后进行在线估计。

ZhaoXiaoyu01, 2022-10-31:
十分感谢

baaixw, 2022-10-31:
我们也会持续改善外参的精度，感谢同学关注我们的工作

DarrenWong, 2022-11-01:
close as no further updates