IPNL-POLYU/UrbanNavDataset #29: IMU direction for VINS-fusiontest

nothing371442, 2022-11-08:
(1) I notice that the direction of imu used in urbanNavDataset is right-front-up. Is it needed to do some transformation to feed the imu data into SLAM algorithm such as Vins-fusion? 
(2) I want to obtain results of vin-fusion， lego-loam and RTK-FGO separately in urbanNavDataset. If I want to do loose coupling, is it a reasonable way to align them by EVO? It belongs to post processing, would it be unfair to do align by such way when real-time operations？
Thank you so much !!

DarrenWong, 2022-11-08:
Thanks for your interest in our dataset.

For (1), yes, you can refer to our [extrinsic](https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/UrbanNav-HK-Deep-Urban-1/extrinsic.yaml) between IMU and camera. You can also refer to the evaluation code [here](https://github.com/baaixw/VINS-Fusion-for-UrbanNavDataset-Evaluation)
and [getting_started](https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/docs/GETTING_STARTED.md)

For (2), do you mean the alignment after getting your results? If you are doing the RPE evaluation, should be fine. However, if you are doing the APE evaluation, alignment is needed to compare the results.

nothing371442, 2022-11-09:
> Thanks for your interest in our dataset.
>
> For (1), yes, you can refer to our [extrinsic](https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/UrbanNav-HK-Deep-Urban-1/extrinsic.yaml) between [...]

Thanks a lot!

closed 2022-11-09 by nothing371442