IPNL-POLYU/UrbanNavDataset #30: about extrinsic

nothing371442, 2022-11-15:
Thanks for your great work!
Has data (orientation and IMU measurements) in imu.csv and reference.csv already been transformed into ENU coordinate and sensor front-left-up coordinate? I notice that it may be different from the description in ROS message and picture, which is NED and sensor right-front-up coordinate. Thanks a lot ！

DarrenWong, 2022-11-16:
Hey, do you mean the imu.csv and reference for non-ros usage in 0428 data? Should be the same as the ROS message. But noted that the IMU is installed as sensor coordinate x- right, y-forward, and z up.

[image]

nothing371442, 2022-11-16:
> Hey, do you mean the imu.csv and reference for non-ros usage in 0428 data? Should be the same as the ROS message. But noted that the IMU is installed as x- right, y-forward, and z up under sensor coordinate
> 
> [image]

Thanks a lot for your kind answer.

I found it that acc from x- right direction has bigger acceleration than y-forward direction when driving begins, where the vehicle is stationary and move slowly. 
[image]

I do IMU dead reckoning using insfilterErrorState method from MATLAB toolbox, and it shows big difference between the result of IMU dead reckoning and the GT from reference.csv. In the picture below, red is GT for 30 s, and blue one is the imu dead reckoning for 30 s.
[image]

I know there may be some problem when I using the MATLAB toolbox. I transformed the IMU input from right-forward-up to MATLAB forward-right-down standard. It did not work. If you have any suggestions, it would be very helpful.

DarrenWong, 2022-11-16:
You can try starting from 25 seconds as the car stop a few time in the beginning in 0428 data.  And might be you can involve the noise term in your algorithm https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/UrbanNav-HK-Data20190428/xsens_imu_param.yaml

Sorry that I'm not familiar with IMU dead reckoning. But I tried a package with dead reckoning, and the results seem reasonable. 
https://github.com/Abekabe/IMU-Dead-Reckoning
[image]

nothing371442, 2022-11-16:
> You can try starting from 25 seconds as the car stop a few time in the beginning in 0428 data. And might be you can involve the noise term in your algorithm [...]

Dear Darren, thank you very much. I will try the package.

closed 2022-11-16 by nothing371442