IPNL-POLYU/UrbanNavDataset #31: About reference conversion

nothing371442, 2022-11-29:
In [UrbanNavDataset](https://github.com/IPNL-POLYU/UrbanNavDataset)/[tools](https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master/tools)/[gt_vis](https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master/tools/gt_vis)/convert_novatel_to_pose.py,
I found it do operation as yaw = np.deg2rad(-1.0 * azimuth), which I am confused about.
I found that azimuth is Eular angle for NED, for it represents the angle between vehicle driving and the North. The yaw in ENU may be defined as the angle between vehicle driving and the East. So, I think it may be yaw = np.deg2rad(90-1.0 * azimuth) to do ENU transformer. I don't know why and look forwards for any suggestions or help.

nothing371442, 2022-11-29:
I also test the velocity in reference.csv, and I found it may be velocity in ENU frame, not in body frame.

DarrenWong, 2022-11-29:
> In [...]

Yes you are right if you need to transform yaw to ENU with **yaw = np.deg2rad(90-1.0 * azimuth)** . The yaw in the convert_novatel_to_pose.py is a middle variable for the transformation only. 
https://github.com/IPNL-POLYU/UrbanNavDataset/blob/73596f3fbe45b24bfd22b9e5b952e27d2ec84c43/tools/gt_vis/convert_novatel_to_pose.py#L27

We apply the transformation and output the angle as rotation matrix/quaternion according to [here](https://hexagondownloads.blob.core.windows.net/public/Novatel/assets/Documents/Bulletins/apn037/apn037.pdf)

https://github.com/IPNL-POLYU/UrbanNavDataset/blob/73596f3fbe45b24bfd22b9e5b952e27d2ec84c43/tools/gt_vis/convert_novatel_to_pose.py#L37

nothing371442, 2022-11-29:
> Yes you are right if you need to transform yaw to ENU with **yaw = np.deg2rad(90-1.0 * azimuth)** . The yaw in the convert_novatel_to_pose.py is a middle variable for the transformation only. [...]

Thanks~ But I notice that in https://hexagondownloads.blob.core.windows.net/public/Novatel/assets/Documents/Bulletins/apn037/apn037.pdf, it said it is needed to convert azimuth to yaw to form 
the matrix. I am wondering why the middle variable is yaw = np.deg2rad(-1.0 * azimuth) rather than yaw = np.deg2rad(90 - 1.0 * azimuth). Because the rotation matrix is typically used to transform right-front-up the ENU.

DarrenWong, 2022-11-29:
> I also test the velocity in reference.csv, and I found it may be velocity in ENU frame, not in body frame.

For your second question, do you mean the velocity in the 0428 data? The header says in the ENU frame. But the rest data, including deep urban, harsh urban etc ,  are represented in the body frame.
Data in 0428
[image]

Data in the GT
[image]

nothing371442, 2022-11-29:
> For your second question, do you mean the velocity in the 0428 data? The header says in the ENU frame. But the rest data, including deep urban, harsh urban etc , are represented in the body frame. [...]

Thanks for your detailed answers, which is very helpful. ^^

DarrenWong, 2022-11-29:
> > > In [UrbanNavDataset](https://github.com/IPNL-POLYU/UrbanNavDataset)/[tools](https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master/tools)/[gt_vis](https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master/tools/gt_vis)/convert_novatel_to_pose.py, I found it do operation as yaw = np.deg2rad(-1.0 * azimuth), which I am confused about. I found that azimuth is Eular angle for NED, for it represents the angle between vehicle driving and the North. The yaw in ENU may be defined as the angle between vehicle driving and the East. So, I think it may be yaw = np.deg2rad(90-1.0 * azimuth) to do ENU transformer. I don't know why and look forwards for any suggestions or help.
> > 
> > 
> > Yes you are right if you need to transform yaw to ENU with **yaw = np.deg2rad(90-1.0 * azimuth)** . The yaw in the convert_novatel_to_pose.py is a middle variable for the transformation only.
> > https://github.com/IPNL-POLYU/UrbanNavDataset/blob/73596f3fbe45b24bfd22b9e5b952e27d2ec84c43/tools/gt_vis/convert_novatel_to_pose.py#L27
> > 
> > We apply the transformation and output the angle as rotation matrix/quaternion according to [here](https://hexagondownloads.blob.core.windows.net/public/Novatel/assets/Documents/Bulletins/apn037/apn037.pdf)
> > https://github.com/IPNL-POLYU/UrbanNavDataset/blob/73596f3fbe45b24bfd22b9e5b952e27d2ec84c43/tools/gt_vis/convert_novatel_to_pose.py#L37
> 
> Thanks~ But I notice that in https://hexagondownloads.blob.core.windows.net/public/Novatel/assets/Documents/Bulletins/apn037/apn037.pdf, it said it is needed to convert azimuth to yaw to form the matrix. I am wondering why the middle variable is yaw = np.deg2rad(-1.0 * azimuth) rather than yaw = np.deg2rad(90 - 1.0 * azimuth). Because the rotation matrix is typically used to transform right-front-up to the ENU.

Sorry for some mistakes in my previous reply. Actually the azimuth = 0 when the vehicle heading north (the azimuth is around y axis, clock-wise), which equals to ENU yaw=0. Therefore, 90 degrees is not necessary.

Page 26 of SPAN CPT manual
[image]

span cpt manual: https://hexagondownloads.blob.core.windows.net/public/Novatel/assets/Documents/Manuals/om-20000122/om-20000122.pdf

nothing371442, 2022-11-29:
> Actually the azimuth = 0 when the vehicle heading north (the azimuth is around y axis, clock-wise), which equals to ENU yaw=0.

I am sorry that, when the azimuth = 0 represents the vehicle heading north (the azimuth is around y axis, clock-wise). Is it equals to ENU yaw=90, and NED yaw = 0 ?

DarrenWong, 2022-11-29:
Do you mean the ENU yaw=90, which x-axis heading north?  I am not very familiar with NED. But it seems to be a transformation rather than only yaw conversion, depending on your definition.

[image]

nothing371442, 2022-11-29:
> Do you mean the ENU yaw=90, which x-axis heading north? I am not very familiar with NED. But it seems to be a transformation rather than only yaw conversion, depending on your definition.
> 
> [image]

I mean when y-axis (vehicle-front direction) heading north, in this situation, ENU yaw=90, at the same time NED yaw=0. For ENU yaw, it may represent the angle between vehicle-front direction and the East. For NED yaw, it may represent the angle between vehicle-front direction and the North.  I think maybe the middle variable "yaw" can use ENU yaw to do transformation.

DarrenWong, 2022-11-29:
Yes yes, you can try to do some.

nothing371442, 2022-11-29:
> Yes yes, you can try to do some.

Thanks a lot, Darren. ^^

closed 2022-11-29 by nothing371442