IPNL-POLYU/UrbanNavDataset #32: EVO对轨迹进行评价出现问题

chenhuahua-orz, 2022-12-08:
您好  我在使用evo对LIO-SAM轨迹与真实轨迹的评价上出现了问题
在未使用轨迹对齐之前 两者的轨迹是这样的
[image]
在使用轨迹对齐 （-a）之后是这样的
[image]
[image]

轨迹明显少了一大截  请问您有什么好办法吗

DarrenWong, 2022-12-11:
你好，lio-sam默认以IMU orientation 初始化第一帧 https://github.com/TixiaoShan/LIO-SAM#prepare-imu-data

如果是单纯评估lidar/imu，建议把 IMU convert 里面的某一行改为单位阵
https://github.com/TixiaoShan/LIO-SAM/blob/4f220164648c53121215a28439bdc2084c50d362/include/utility.h#L268

此外，出现轨迹少一截因为lidar时间戳跟真值不是完全对应，evo align后需要添加` --t_max_diff` 参数align真值到最近的lidar，或者把真值插值到lidar对应时间戳。
https://github.com/TixiaoShan/LIO-SAM/blob/4f220164648c53121215a28439bdc2084c50d362/include/utility.h#L268

yinhaosen, 2023-04-03:
想问一下大神用GPS作为真值，如何转换GPS的原始数据变成EVO可以用来评估的数据？

DarrenWong, 2023-04-11:
你可以加一个sub pub 把novatel_msgs/INSPVAX转成NatSatFix。可以验证GPS加入的有效性。

但是一般很少直接融合真值，会采用一些车载设备(例如Ublox M8T和F9P)的经纬度信息融合进去，可以参考我们之前的研究:https://www.mdpi.com/2076-3417/12/10/5193

ajie1231, 2023-12-14:
> 想问一下大神用GPS作为真值，如何转换GPS的原始数据变成EVO可以用来评估的数据？

老哥您好，可以私聊您么，最近也在使用liosam跑数据集，有一些问题想向您请教

ajie1231, 2023-12-14:
> > > 想问一下大神用GPS作为真值，如何转换GPS的原始数据变成EVO可以用来评估的数据？
> > 
> > 
> > 你可以参考下这里https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master/tools/gt_vis
> 
> 我想测试LIO-SAM加上你们数据集中GPS信息后的效果，但是我看LIO-SAM源代码本身使用的是NatSatFix类型的GPS数据，你们数据集中GPS数据是用的novatel_msgs/INSPVAX，怎么转换进LIO-SAM呢？

老哥您好，可以私聊您么，最近也在使用liosam跑数据集，有一些问题想向您请教

yinhaosen, 2023-12-19:
> > > > 想问一下大神用GPS作为真值，如何转换GPS的原始数据变成EVO可以用来评估的数据？
> > > 
> > > 
> > > 你可以参考下这里https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master/tools/gt_vis
> > 
> > 
> > 我想测试LIO-SAM加上你们数据集中GPS信息后的效果，但是我看LIO-SAM源代码本身使用的是NatSatFix类型的GPS数据，你们数据集中GPS数据是用的novatel_msgs/INSPVAX，怎么转换进LIO-SAM呢？
> 
> 老哥您好，可以私聊您么，最近也在使用liosam跑数据集，有一些问题想向您请教

可