IPNL-POLYU/UrbanNavDataset #33: Question about the ls lidar channel

minwoo0611, 2022-12-22:
Hello. Thanks for the providing the open source dataset. I'm really interested on the UrbanNavDataset, and checking the SLAM performance with it.

I haven't check the all the dataset, but in the case of UrbanNav-HK-Harssh, I found that the lslidar has the time channel from 0.0s to 0.2s. As the topic of lslidar is 10Hz, it seems that time channel should be 0.0s to 0.1s. Can you explain the reason of this situation or give me some suggestions?

Thank you.

minwoo0611, 2022-12-23:
It seems I made a mistake. In fact, there is no time channel at all.

Sorry for the mistake and I closed the issue.

closed 2022-12-23 by minwoo0611

DarrenWong, 2022-12-23:
Yes, the [lslidar driver](https://github.com/IPNL-POLYU/lslidar_C16) does not provide a time channel. You can refer to the right slant Velodyne if you require the time information.