IPNL-POLYU/UrbanNavDataset #34: Extrinsics of IMU and GPS

RajatArora-CA, 2022-12-30:
First of all I would like to Thank you for such a great dataset.

I had the following questions related to extrinsic:
1. IMU frame mentioned in extrinsic parameters file of each sequence is of  Xsens MTi 10 or Phone IMU: Xiaomi 8, 239Hz?
If it is of Xsense MTi 10 then what is the extrinsic to the Phone IMU frame?
3. In the GNSS data of each sequence, there are nmea files from Ublox m8t and f9p and some mobile phones. Is the extrinsic parameters of Ublox m8t and f9p the same which is mentioned as Extrinsic parameter from GNSS antennas to IMU in the extrinsic parameters file of each sequence. If not please provide their extrinsic.
4. Please provide the extrinsic corresponding to mobile phones from which the GNSS data is available in the form of nmea files.

DarrenWong, 2022-12-30:
Hi, thanks for your interest in our dataset.

 For Q1, it is Xsense MTi 10.  We do not have an accurate extrinsic for the Phone IMU. It is mounted on the top middle of the front windshield. Its initial translation can be approximate to the GNSS antenna attached to the car front. Then you might optimize it by yourself. https://github.com/IPNL-POLYU/UrbanNavDataset/blob/73596f3fbe45b24bfd22b9e5b952e27d2ec84c43/UrbanNav-HK-Deep-Urban-1/extrinsic.yaml#L60

For Q2, yes we use the same extrinsic as we place it very close to each other which is mounted on the vehicle front. But another F9P is connected with a geodetic antenna in the car rear, which is named "xxxx f9p.splitterxxx",  the extrinsic from xsense IMU to GNSS is:
             x=0.000, y=-0.560, z=-0.210 m (x-right, y-fwd, z-up)

For Q3, do you mean the extrinsic among the mobile phones? We place the phones horizontally in the vehicle front with an interval of around 2-3cm.

RajatArora-CA, 2022-12-30:
Thank you for the answer.
I had one more doubt regarding the phone IMU data.
[image]
The above snippet is from the csv file of phone IMU.

ivannhf, 2023-01-03:
Hi, the sensor framework should follow the standard coordinate system in the Android development page (https://developer.android.com/guide/topics/sensors/sensors_overview), as shown follows, 
[image]

The Xiaomi Mi 8 is rotated in landscape mode for data recording. Therefore, the x-axis points downward, and the -ve z-axis is aligned with the driving direction.

RajatArora-CA, 2023-01-03:
Thank you for the answer.

closed 2023-01-03 by RajatArora-CA