IPNL-POLYU/UrbanNavDataset #35: External parameter question about zed camera and imu

zhanglei1997, 2023-01-03:
According to the sensor installation diagram provided, the installation positions of IMU and zed are very close, but in the external reference document provided by the official:
################## Extrinsic parameter between IMU and Camera##################
################## camera is ZED2, a stereo camera##################
LEFT_CAMERA_T_IMU:!!opencv-matrix (Left camera to IMU)
 rows: 4
   cols: 4
   dt: d
   data:   [ 9.9885234402635936e-01, 1.3591158885981787e-03,
       4.7876378696062108e-02, -8.4994249456545504e-02,
       -4.7864188349269129e-02, -7.9091258538426246e-03,
       9.9882253939420773e-01, 6.6169337079143220e-01,
       1.7361758877140372e-03, -9.9996779874765440e-01,
       -7.8349959194297103e-03, -3.0104266183335913e+00, 0., 0., 0., 1. ]

The translation part of the 4x4 matrix, z direction is -3.0104266183335913e+00. A translation parameter larger than 3m is obviously wrong because the camera should not be located 3m below the IMU. I would like to know whether I used the parameter incorrectly or there is a problem with the external parameter calibration?

DarrenWong, 2023-01-04:
Hi you can refer to the issue #17  and #27

zhanglei1997, 2023-01-04:
Hello. These problems have corrected the translation in the other two directions, but the translation in the z direction of 3m has not been mentioned. It is hoped that the external parameters of translation can be corrected. Thank you very much。

DarrenWong, 2023-01-04:
The calibration results of the z-direction of the left camera ZED2 is not good so far. You can use some online calibration tools (e.g., VINS mono) to have a better extrinsic parameter.

DarrenWong, 2023-01-25:
closed as no further updates

closed 2023-01-25 by DarrenWong