IPNL-POLYU/UrbanNavDataset #38: pseudo-distance was much larger than the real distance

KitKator, 2023-03-23:
I used the gnss-deep data set in Hong Kong and found that the pseudo-distance was much larger than the real distance, in which the pseudo-distance had been processed by satellite ephemeris error and ionospheric tropospheric error. The real distance was the distance from the satellite to the receiver, and the real position and obs observation files you provided were used. May I ask whether there is a deviation of about 0.005s in the time unification of obs file?
I used 20210521.medium-urban.whampoa.ublox.f9p.obs and UrbanNav_whampoa_raw.txt

ivannhf, 2023-03-23:
Thank you for using the dataset. 
Have the satellite and receiver clock offsets been compensated during the evaluation of the pseudorange observation?

KitKator, 2023-04-03:
The receiver clock error has been eliminated, thank you very much for your help

DarrenWong, 2023-04-07:
closed as fixed

closed 2023-04-07 by DarrenWong