IPNL-POLYU/UrbanNavDataset #41: The standard deviation of GNSS latitude and longitude heights in the UrbanNav-HK-Deep Urban-1 file being 0

JoyuZheng, 2023-04-17:
Hello, thank you for the excellent work of your team. During my use of this dataset, I read the topic data of/novated_data/inspvax in the "UrbanNav-HK-Deep Urban-1" bag file, as shown in the following figure. The standard deviation of longitude and latitude height is 0. Have you provided it elsewhere?
[image]

DarrenWong, 2023-04-18:
We do not extract the standard deviation of the GNSS/INS solution as it is the ground truth positioning in our dataset. But it has a quality indicator to evaluate the accuracy of the ground truth  (the postioning_type in the message) https://www.dropbox.com/s/twsvwftucoytfpc/UrbanNav_TST_GT_raw.txt?dl=0
Reference: 
https://docs.novatel.com/Waypoint/Content/PDFs/Waypoint_Software_User_Manual_OM-20000166.pdf
[image]

JoyuZheng, 2023-04-18:
I see. All right. Thank you for taking the time to reply to me. I have another question, which is that in the dataset "UrbanNav-HK-Data20190428", when I converted the GNSS data of the topic "/novated_data/inspvax" into data of type "sensor_msgs/NavSatFix" in the following way, I found an offset between the converted timestamp and the "bag timestamp". The offset is about a few seconds. May I ask what is the reason for this deviation? How can I align the converted timestamp with bag time?Thank you!!!
conversion method：
navsatfix_msg.header.stamp = rospy.Time.from_sec((msg.header.gps_week * 604800.0 + msg.header.gps_week_seconds/1000.0-18) + 315964800)

DarrenWong, 2023-04-18:
Your time conversion formula is correct. But noted that the 0428 data is not time synced with GNSS https://github.com/IPNL-POLYU/UrbanNavDataset/issues/39. The time offset is around 2.xx seconds. You might use the [medium urban data](https://github.com/IPNL-POLYU/UrbanNavDataset#urbannav-hk-medium-urban-1) instead if you want to fully align the GNSS and the ROS timestamp

JoyuZheng, 2023-04-18:
Okay, thank you for the answer. The main reason I used the 0428 data is because, as mentioned earlier, the standard deviation is not included in the 2021 dataset.

closed 2023-04-20 by JoyuZheng