IPNL-POLYU/UrbanNavDataset #42: Problem when applying dataset 20190428 to VINS-Mono

NeF51, 2023-04-19:
Hello, thank you for your work.
I have experienced a problem when applying dataset 20190428 to VINS-Mono. The estimated system frequently reboot and the estimated trajectory comes back to the original point. Just as the image illustrated as follow.
[image]

Have you ever experienced the same problem? And I would be very grateful if you know how to solve it.

NeF51, 2023-04-19:
Additionally, this problem usually happened when the car stopped for a while.

baaixw, 2023-04-20:
Hi, I am not sure from the above contents you posted, do you open the loop closure? If yes, can you just close it, then try again.

baaixw, 2023-04-20:
> Additionally, this problem usually happened when the car stopped for a while.

Hi, I am not sure from the above contents you posted, do you open the loop closure? If yes, can you just close it, then try again. Actually, when the car stops for too long, the possibility of drifting increases.

NeF51, 2023-04-20:
> 

Thank you for your reply. I have tried closing the loop closure but the problem still exists. 
In the optimization codes of VINS-Mono, I found that the system will reboot when it detect nonlinear optimization failure. Thus, I assume that the reason might be that the long time stops of the car caused drifting error of IMU, which led to the failure of nonlinear optimization and the reboot of the system. 
Then, to verify my assumption, I tried dataset 20200314 on VINS-Mono. The car in this dataset hardly stops. And the estimation turned out to be fine, where the system only reboot once (When car slowed down and turned a very sharp bend).  
Is it mean that VINS-Mono might not suitable on dataset 20190428?

NeF51, 2023-04-20:
> > Additionally, this problem usually happened when the car stopped for a while.
> 
> Hi, I am not sure from the above contents you posted, do you open the loop closure? If yes, can you just close it, then try again. Actually, when the car stops for too long, the possibility of drifting increases.

Thank you for your reply. I have tried closing the loop closure but the problem still exists.
In the optimization codes of VINS-Mono, I found that the system will reboot when it detect nonlinear optimization failure. Thus, I assume that the reason might be that the long time stops of the car caused drifting error of IMU, which led to the failure of nonlinear optimization and the reboot of the system.
Then, to verify my assumption, I tried dataset 20200314 on VINS-Mono. The car in this dataset hardly stops. And the estimation turned out to be fine, where the system only reboot once (When car slowed down and turned a very sharp bend).
Is it mean that VINS-Mono might not suitable on dataset 20190428?

baaixw, 2023-04-20:
> Thank you for your reply. I have tried closing the loop closure but the problem still exists. In the optimization codes of VINS-Mono, I found that the system will reboot when it detect nonlinear [...]

I don't think so. There maybe other factors will lead to the bad result, e.g. the initialization, the device ... My suggestion is that:
1. When you play the rosbag, slow the speed ;
2. The initialization is not very stable.  If the vehicle stops for too long, you can try to skip a few seconds at the beginning

Actually, this data 20190428 is inherently challenging. We can keep discussion on the reason behind the drift.

closed 2023-04-23 by NeF51