IPNL-POLYU/UrbanNavDataset #43: Confusion about the direction of angles

NeF51, 2023-04-23:
Dear authors,
  Thank you for your work. Recently, I am working on IMU dead reckong with dataset 20190428 imu.csv and reference.csv. And I have some questions about the direction of the angular rate and angle in these files.
  Which type of direction of angular rate and heading angle in the following picture is correct in imu.csv and reference.csv? The heading angle increases when the vehicle turning anticlockwise or clockwise? Is the forward direction of angular rate of Z axis correct (anticlockwise)?
[image]

DarrenWong, 2023-04-24:
Yes, the direction is the same as your right picture illustration. 
 The heading angle in _reference.csv_ increases when the vehicle turns clockwise according to the ground truth positioning provided by the [Novatel Azimuth](https://docs.novatel.com/OEM7/Content/SPAN_Logs/INSPVAX.htm)
The starting point heading in _reference.csv_ is 228 degrees as shown below
[image]

The forward direction of the angular rate of Z axis in the _imu.csv_ is anticlockwise following the standard [ROS frame](https://www.ros.org/reps/rep-0145.html).  All the frame is right-handed.

NeF51, 2023-04-24:
> Yes, the direction is the same as your right picture illustration. The heading angle in _reference.csv_ increases when the vehicle turns clockwise according to the ground truth positioning provided [...]

Thanks!

closed 2023-04-24 by NeF51