IPNL-POLYU/UrbanNavDataset #44: Tunnel Dataset issue

jcmayoral, 2023-06-23:
I have tried to download the dataset with the tunnel. However, once downloaded the bag has some issues and just deployed the initial 57 seconds of the video and stops once the vehicle starts entering the tunnel. Am I doing something wrong?
[image]

DarrenWong, 2023-06-23:
Hi, I think you didn't download the whole rosbag as you rosbag only 2GB and reindex is required. The whole rosbag should be around 16.8GB

[image]

[image]

jcmayoral, 2023-06-23:
It is weird. Because before I reindex its 16.8GB and the reindexed bag is 2gb.

DarrenWong, 2023-07-14:
closed as no further updates

closed 2023-07-14 by DarrenWong