IPNL-POLYU/UrbanNavDataset #48: Skymask

MrTHMX, 2023-11-15:
Thank you for your work on the dataset!
Can you provide some help on how to use the skymask?

DarrenWong, 2023-11-16:
Hi the skymask is the mask degree from 0-360 degrees of a specific region, which can be downloaded [here](https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master#datasets).
You can query the correspondence Skymask (as shown in the excel files, from the skymask item) based on your current locations. 

```
Latitude(deg)	Longitude(deg)	Altitude(m)	Skymask(1*361	deg)		
22.29566762	114.1730411	5	82	82	82	82
22.29568568	114.1730411	5	83	83	83	83
22.29570374	114.1730411	5	85	85	85	85
22.2957218	114.1730411	5	87	87	87	87
22.29610109	114.1730411	5	89	89	89	89
22.29667905	114.1730411	5	89	89	89	89
```

MrTHMX, 2023-11-16:
The position in the ground truth file does not correspond well to the position in the skymask. What should I do?

DarrenWong, 2023-11-16:
Yes, we compute the skymask of the corresponding scenarios based on the 3D model. You can use the nearest point compared to your ground truth in the skymask files as the reference skymask

MrTHMX, 2023-11-17:
Is there a way to get a more precise position of the skymask, such as interpolation? Can you give me some advice?

DarrenWong, 2023-11-17:
I am not trying interpolation based on the skymask yet. But the edge of the skymask might not change too much between two consecutive points in the skymask files.

MrTHMX, 2023-11-19:
Thank you very much for your help！

closed 2023-11-19 by MrTHMX