IPNL-POLYU/UrbanNavDataset #50: data synchronization

Mahmoudadham4040, 2023-12-05:
could you please tell me which Hong Kong dataset is synchronized for all sensors and also with ground truth?

DarrenWong, 2023-12-05:
The data [UrbanNav-HK-Medium-Urban-1](https://github.com/IPNL-POLYU/UrbanNavDataset#urbannav-hk-medium-urban-1), [UrbanNav-HK-Deep-Urban-1](https://github.com/IPNL-POLYU/UrbanNavDataset#urbannav-hk-deep-urban-1), [UrbanNav-HK-Harsh-Urban-1](https://github.com/IPNL-POLYU/UrbanNavDataset#urbannav-hk-harsh-urban-1) and [UrbanNav-HK-Tunnel-1](https://github.com/IPNL-POLYU/UrbanNavDataset#urbannav-hk-tunnel-1) synced the ROS system with the GPS time, also the ground truth

Mahmoudadham4040, 2023-12-07:
Thanks for your reply and consideration, 
For the UrbanNav-HK-Medium-Urban-1, I checked the time of integrated sensor inside the rosbag with the GNSS files (F9p.splliter) but I see it is not synchronized exactly, is it true or there are something wrong?

DarrenWong, 2023-12-14:
closed as no further updates

closed 2023-12-14 by DarrenWong