IPNL-POLYU/UrbanNavDataset #51: LIOSAM测试

ajie1231, 2023-12-14:
大佬您好 我想请教一下我如果想试试liosam跑HK-Medium-Urban-1数据集，只用到lidar和imu数据 需要怎么调整外参 期待您的回复 谢谢！

DarrenWong, 2023-12-15:
你好 我们有modified过的liosam外参直接可以跑的，你可以通过下面链接获取
https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/docs/GETTING_STARTED.md#lidar-inertial-odometry-video

ajie1231, 2023-12-15:
> 你好 我们有modified过的liosam外参直接可以跑的，你可以通过下面链接获取 https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/docs/GETTING_STARTED.md#lidar-inertial-odometry-video

收到 十分感谢您！！！！！

closed 2023-12-19 by DarrenWong

ajie1231, 2023-12-27:
老师您好 我跑了TST之后发现z轴会有漂移 请问两个配置文件params.yaml以及params_velodyne.yaml 应该用的哪一个呢 这两个是对应什么情况下的

DarrenWong, 2023-12-28:
你好，请使用params_velodyne.yaml.   params.yaml是liosam默认数据集的yaml，没有对这个进行修改。z轴会有漂移指的是有z轴高度差对吗？这个其实是一个比较LiDAR里程计比较常见的情况，我们之前的研究做过一些相关的研究可以供参考,
1.Wen W, Hsu L T. AGPC-SLAM: Absolute ground plane constrained 3D LiDAR SLAM[J]. NAVIGATION: Journal of the Institute of Navigation, 2022, 69(3), https://navi.ion.org/content/69/3/navi.527.abstract
2. F. Huang, W. Wen, J. Zhang and L. -T. Hsu, "Point Wise or Feature Wise? A Benchmark Comparison of Publicly Available Lidar Odometry Algorithms in Urban Canyons," in IEEE Intelligent Transportation Systems Magazine, vol. 14, no. 6, pp. 155-173, Nov.-Dec. 2022, doi: 10.1109/MITS.2021.3092731., https://ieeexplore.ieee.org/document/9748891

ajie1231, 2024-01-03:
谢谢您的回复，2024新年快乐！那我去拜读一下您的论文 ，还有一个问题就是想请问一下Medium-Urban1和Tunnel是一个IMU内外参嘛，在入隧道后算法定位就漂了，完全失效了。

DarrenWong, 2024-01-03:
你好，IMU内外参都是一样的，基本都是同一两天采集的，隧道主要是因为feature约束不够，LiDAR退化导致失效。这个是一个很好的研究点，如何用LIO跑出来这种特征比较相似的隧道。

有一篇论文有做过这几个场景的LIO评估，供你参考，https://www.mdpi.com/2072-4292/14/12/2835