IPNL-POLYU/UrbanNavDataset #55: timestamp of SPAN-CPT in UrbanNav-HK-Data20200314 is not match with te image

Shujie-Hu, 2024-01-17:
I extracted the images in the bag and the trajectory of SPAN-CPT with timestamp, but the motion state of the vehicle does not match the trajectory of SPAN-CPT.
[image]
SPAN-CPT trajectory show vehicle is turning at gpsWIS 549992.0
[image]
but image show vehicle is turning at gpsWIS 550010
Could you please tell me why this problem occurs?

DarrenWong, 2024-01-19:
Hello, can I know the UTC timestamp of the vehicle image? As the UTC and GPS time have 18 leaf seconds, it might possible have some problem when you compare the time directly. You can refer to this [page](https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/docs/GETTING_STARTED.md#time-synchronization-between-gps-time-and-ros-time) for the conversion.

Shujie-Hu, 2024-01-20:
I have reconfirmed the code for extracting images from the bag, and I think that the timestamp of the image message header is UTC time and converted it to GPS time. Is there a problem with this process? I will take a screenshot of the code as follows:
[image]
[image]

DarrenWong, 2024-01-20:
Hey Shujie, your code is correct. Seems the rosbag timestamp indeed has a gap with the GPS time as no hardware time sync was applied in the pilot data. Suggest you can use the data collected in 2021 or refer to the timestamp of rosbag topic  between `/novatel_data/inspvax` and `/camera/image_color/header` to adjust the time manual. 

The UTC stamp I tested was based on Matlab [function](https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/docs/GETTING_STARTED.md#time-synchronization-between-gps-time-and-ros-time) and https://www.labsat.co.uk/index.php/en/gps-time-calculator.
1584175590 (UTC)- > week: 2096, second: 550010 (GPS)   -> the rosbag is (549993) at the same time.

Shujie-Hu, 2024-01-21:
> Hey Shujie, your code is correct. Seems the rosbag timestamp indeed has a gap with the GPS time as no hardware time sync was applied in the pilot data. Suggest you can use the data collected in 2021 [...]

Thanks for your response! I'll try again.

closed 2024-01-22 by DarrenWong