IPNL-POLYU/UrbanNavDataset #56: Distortion occurs when comparing with the true trajectory

HowieHepburn, 2024-03-16:
You can find the error discription here. 
https://github.com/MichaelGrupp/evo/issues/643#issue-2189861160
I am wondering if this is normal and whether the error estimation results obtained by EVO tool  is trustworth or not

DarrenWong, 2024-03-17:
As the provided ground truth is 1 hz that might be some point is not associated with the sensor timestamp,  you can try using _--t_max_diff 0.1_ to find the nearest timestamp if the estimated trajectory is 10 hz.

e.g,

evo_ape tum gt_slam_UrbanNav-HK-TST-20210517.csv lio_odom_UrbanTST.tum -pa --t_max_diff 0.1

And as for your trajectory, it seems not to start from (0,0), did you adjust some original points?

HowieHepburn, 2024-03-18:
> As the provided ground truth is 1 hz that might be some point is not associated with the sensor timestamp, you can try using _--t_max_diff 0.1_ to find the nearest timestamp if the estimated [...]

Thank you so much for your reply, Your suggestion to incorporate the "--t_max_diff 0.1" parameter in evaluating trajectories has proven to be immensely helpful. 
evo_ape tum gt_slam_UrbanNav-HK-TST-20210517.csv lio_odom_UrbanTST.tum -pa --t_max_diff 0.1:
[image]

Regarding your observation about the trajectory not starting from (0,0),  I didn't realize I adjusted some of the original points,I believe the discrepancy may stem from the method used to generate trajectory points. My trajectory points were extracted from a rosbag and subsequently saved in TUM format. This rosbag was recorded manually, Perhaps this process might have introduced slight deviations from the original data, leading to the observed discrepancy?

Thank you once again for your support.

DarrenWong, 2024-03-25:
closed as completed

closed 2024-03-25 by DarrenWong