IPNL-POLYU/UrbanNavDataset #57: ERROR bag unindexed: 2019-04-28-20-58-02.bag(UrbanNav-HK-Data20190428)

HowieHepburn, 2024-05-17:
When I get the UrbanNav-HK-Data20190428 dataset via dropbox link (https://www.dropbox.com/s/pfjvauvletdpnh2/2019-04-28-20-58-02.bag?e=1&dl=0)
-:rosbag info 2019-04-28-20-58-02.bag 
-:ERROR bag unindexed: 2019-04-28-20-58-02.bag.  Run rosbag reindex.
-:rosbag reindex 2019-04-28-20-58-02.bag 
 2019-04-28-20-58-02.bag                     100%             13.8 GB 00:09
 -:rosbag info 2019-04-28-20-58-02.bag
 -:
 .....
 duration:    2:45s (165s)
 .....
And the duration should be 387s from your description, I wonder what's wrong with it and how should I get the correct dataset

DarrenWong, 2024-05-17:
Thank you for your interest in our dataset. Can I know if you download the whole file successfully? It should be around 40GB. It might be have some Interruption during the your download process. Can you try download it again using Edge or firefox or Dropbox Desktop?

HowieHepburn, 2024-05-17:
Thank you for your prompt reply, and you are right, I didn't download of the whole file, but chrome showed that the download was completed.
Now I downloaded two compressed packages through Baidunetworkdisk . And I noticed that it has two parts. How to use them?
And what's the diffrent between [UrbanNav-HK-Medium-Urban-1] and [(Pilot data) UrbanNav-HK-Data20190428]?Is it just the difference between one lap and two laps?

DarrenWong, 2024-05-17:
[UrbanNav-HK-Medium-Urban-1] and [(Pilot data) UrbanNav-HK-Data20190428] are two collected data in the same place. suggest you can use the [UrbanNav-HK-Medium-Urban-1] as it is the updated version with multiple GNSS receiver.

HowieHepburn, 2024-05-17:
Thanks a lot for your reply.

closed 2024-05-17 by HowieHepburn