IPNL-POLYU/UrbanNavDataset #59: Have some problem with the dataset.

hello299, 2024-06-14:
Hello,I am a student,I am sorry for that I want to use chinese to cleary express my question:
  您好，目前我的研究利用到harsh urban 中的IMU数据，但是我观察到Harsh urban 中的数据集同样在sensor_msgs中存在IMU的数据，这部分数据和您提供的IMU数据是否一样？因为我的研究没有用到vision以及lidar，我想知道ROS包中存在IMU数据集是否与您单独提供的IMU数据是一致的，同时如果数据不一致，您方便发我一份ROSbag中的IMU数据么？因为rosbag数据包很大，导致下载很慢
如果我没有清晰的表述我的意思，您可以加我的微信。
谢谢老师您的回复

DarrenWong, 2024-06-14:
谢谢对我们数据集的关注，单独提供的IMU数据和ROS包里面是一致的，单独提供是为了方便非ROS用户使用。

hello299, 2024-06-14:
好的，谢谢老师，相关的IMU参数的配置从yaml文件中获得，不知道我理解的是否正确，同时，我也出现了在ubuntu系统20.04运行ROS的问题，当输入sudo apt-get install ros-noetic-nav-msgs出现报错的情况，显示无法定位软件源，我去了ROS官网，发现20.0只是提供了ome7-driver 以及gps-driver，想了解一下，这个问题的处理，谢谢老师指导

hello299, 2024-06-14:
老师，也同时想请教您一下，在[UrbanNav-HK-Harsh-Urban-1](https://github.com/IPNL-POLYU/UrbanNavDataset#urbannav-hk-harsh-urban-1)数据集中，存在xsense以及xsense_update文件，这两个文件，是相互补充？还是update是未加update的文件的替代？

DarrenWong, 2024-06-14:
> 老师，也同时想请教您一下，在[UrbanNav-HK-Harsh-Urban-1](https://github.com/IPNL-POLYU/UrbanNavDataset#urbannav-hk-harsh-urban-1)的IMU数据集中，存在xsense以及xsense_update文件，这两个文件，是相互补充？还是update是未加update的文件的替代？

xsense文件是直接从ROS导出，后面为了简洁一些，我们就加了一个xsense_update, 两个数据是一致的，只是xsense_update简化了header

hello299, 2024-06-15:
真的十分的感谢您，老师，解答了我的两个疑惑！实在麻烦您抽出百忙时间帮我。（上面两个问题已经解决）
老师，针对于一些IMU数据的问题，我存在几点疑惑，方便向老师请教一下：因为，我之前自己实现松组合的时候，考虑到了一些IMU的参数，老师的IMU数据集这些参数我可能粗心，只是找到了数据频率以及陀螺和加速度计的噪声和随机游走，老师方便提供一些诸如：初始的速度信息（北东地）、初始行角度信息（row pitch heading ）,IMU初始的误差（陀螺仪以及加速度计的零偏），安装参数（天线杆臂）以及IMU到载体的旋转角（安装角），这是一些我考虑到的参数，麻烦老师了，向老师请教一下～

DarrenWong, 2024-06-15:
初始的速度信息（北东地）、初始行角度信息（row pitch heading ）可以考虑用真值获取.IMU初始的误差（陀螺仪以及加速度计的零偏）这个暂时我们没有提供哦，是否也可以考虑跟Novatel提供的真值的roll pitch yaw对齐？
安装参数（天线杆臂）以及IMU到载体的旋转角（安装角）可以参考这个[文件](https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/UrbanNav-HK-Medium-Urban-1/extrinsic.yaml)，但是目前是manual量出来的，精确度10厘米左右

```
################## Extrinsic parameter from GNSS antennas to IMU ##################
# only translation part are measured manually (±0.1), y-axis forward, x-axis right and z-axis up
ANTENNA_T_IMU: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 1, 0, 0, 0,
           0, 1, 0, 0.86,
           0, 0, 1, -0.31,
           0, 0, 0, 1]
```

hello299, 2024-06-15:
[image]
[image]
老师，这两个数据，在gt真值中，对应的velBdx以及AccBdx我可以理解成为初始的速度以及航向信息么？
同时，我不太明白，为什么IMU数据中存在orientation:x,y,z,w四个方向 而真值中仅仅存在三个方向？w方向应该如何对应？

hello299, 2024-06-15:
################## Extrinsic parameter from GNSS antennas to IMU ##################
# only translation part are measured manually (±0.1), y-axis forward, x-axis right and z-axis up
ANTENNA_T_IMU: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 1, 0, 0, 0,
           0, 1, 0, 0.86,
           0, 0, 1, -0.31,
           0, 0, 0, 1]我能再问一下老师，这个为什么 是四维？我是以三维向量模式传入的 那么方便老师指导一下怎么把四维矩阵转换得到三维我需要的东北天向的安装角度么？我想利用的是三维向量的形式 所以不知道需要进行怎样的转换

DarrenWong, 2024-06-15:
这是旋转矩阵形式而已，如果你需要三位向量，你就采用矩阵第四列的 [0 0.86 -0.31] 对应上x y和z，GNSS天线在IMU的y轴的0.86米

hello299, 2024-06-16:
好的老师，还有这个问题，麻烦老师解答一下，谢谢老师。还有上面的问题，麻烦老师解答一下：为什么IMU数据中存在orientation:x,y,z,w四个方向 而真值中仅仅存在三个方向？w方向应该如何对应？https://github.com/IPNL-POLYU/UrbanNavDataset/issues/59#issuecomment-2169212276

DarrenWong, 2024-06-17:
> 好的老师，还有这个问题，麻烦老师解答一下，谢谢老师。还有上面的问题，麻烦老师解答一下：为什么IMU数据中存在orientation:x,y,z,w四个方向 而真值中仅仅存在三个方向？w方向应该如何对应？[#59 (comment)](https://github.com/IPNL-POLYU/UrbanNavDataset/issues/59#issuecomment-2169212276)

你好 IMU ROS里面是四元数表示，可以参考这个的转换到你想到的形式，https://www.andre-gaschler.com/rotationconverter/
真值的是roll pitch yaw形式，参考novatel的格式的item11-13， https://docs.novatel.com/OEM7/Content/SPAN_Logs/INSPVAX.htm

关于真值转换为TUM格式，可以使用我们的一个工具，https://github.com/IPNL-POLYU/UrbanNavDataset/tree/master/tools/gt_vis

hello299, 2024-06-17:
好的谢谢老师，问题已经解决，我想问一下 对于IU噪声建模 相关时间(correlation time ），针对于harsh urban的数据应该如何进行恰当的选择，噪声建模选择一阶高斯马尔科夫过程。

DarrenWong, 2024-06-17:
你可以先试试用一阶高斯马尔科夫过程是否适合你这边的软件跑我们这笔数据。但是我们这是车载级别的IMU，噪声比较大，你松耦合的话可以考虑放大一些标定好的噪声，这样跑起来效果可能会更好一些

hello299, 2024-06-17:
车载级别的IMU，老师您推荐什么模型进行IMU噪声建模？
没太明白老师说放大标定好的噪声是什么意思？这些标定好的噪声对于您这个IMU设定的参数信息我需要怎么获取？
同时，老师，在这个imu误差估计中，需要提供外部的IMU陀螺仪以及加速度计的偏差标准差，我在老师的参数文件中，可能是我的疏忽，并没有搜寻到
[image]
计中，需要提供外部的IMU陀螺仪以及加速度计的偏差标准差，我在老师的参数文件中，可能是我的疏忽，并没有搜寻到

hello299, 2024-06-17:
我不知道我想的是否正确，IMU标定好的噪声，是应该针对于老师这款IMU提供的参数信息是么？我应该如何获取？

DarrenWong, 2024-06-17:
> 车载级别的IMU，老师您推荐什么模型进行IMU噪声建模？ 没太明白老师说放大标定好的噪声是什么意思？这些标定好的噪声对于您这个IMU设定的参数信息我需要怎么获取？ 同时，老师，在这个imu误差估计中，需要提供外部的IMU陀螺仪以及加速度计的偏差标准差，我在老师的参数文件中，可能是我的疏忽，并没有搜寻到，老师方便提供以下思路么 ![2024-06-17 20-33-36 的屏幕截图](https://private-user-images.githubusercontent.com/74706451/340306447-b8d22323-e006-42ee-9508-cbd34b84b730.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2MzIzNTMsIm5iZiI6MTcxODYzMjA1MywicGF0aCI6Ii83NDcwNjQ1MS8zNDAzMDY0NDctYjhkMjIzMjMtZTAwNi00MmVlLTk1MDgtY2JkMzRiODRiNzMwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjE3VDEzNDczM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEyYTdmNjM5OWVhNmQzYzdlNGZjYjI2MGI2YWRkMWEyYjRlZDE3MjExZmY5MzAzYTRmZDNiOTY0NTNkYjQwZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.b1N6G3-wzjq5NgNFacz1P-TFCWz2Fbz4y-2m5vf6F8U)

你好，我们目前标定的IMU误差模型包括了Random Walk 和 White noise，可以参考这个文件下的[数据](https://github.com/IPNL-POLYU/UrbanNavDataset/blob/master/UrbanNav-HK-Harsh-Urban-1/xsens_imu_param.yaml)， 具体标定的过程，你可以参考我们用的一个开源工具 [imu_utils](https://github.com/gaowenliang/imu_utils) 和相关[说明](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model#how-to-obtain-the-parameters-for-your-imu). 

另外你从datasheet找到的参数也是可以的，先放进去测试是否可行。

hello299, 2024-06-18:
谢谢老师的耐心答疑，有问题在向老师请教

hello299, 2024-06-18:
https://github.com/IPNL-POLYU/UrbanNavDataset/issues/59#issuecomment-2173281723
老师，在这里，我通过您上面的提示，已经将parameter里面的acc以及gry的arw以及noise设置为了相应的文件中的信息，可能我没有表示清楚我的问题，实现中 还需要提供IMU设备的陀螺仪以及加速度计的bias的std标准差数据 基于代码作者的设置如下：
[image]

DarrenWong, 2024-06-22:
> 老师，那我方便在向您请教一下，[https://github.com/IPNL-POLYU/UrbanNavDataset/issues/59#issuecomment-2173085239，](https://github.com/IPNL-POLYU/UrbanNavDataset/issues/59#issuecomment-2173085239%EF%BC%8C)
> 
> 随机游走以及白噪声是提前标定的IMU误差，您在上面这个连接中，说放大标定好的噪声，这个具体能提示个思路怎么放大？或者老师举个简单的例子，我根据这个数据举一反三一下？

就是实际场景中，假如IMU现有的标定噪声参数上，你可以再增加5-10倍，例如如果噪声是0.00001，你可以尝试0.0001，更相信GNSS的结果，来对比下是否对算法有所改善。

DarrenWong, 2024-06-22:
具体你可以参考这款IMU的datasheet和你这个代码的定义，例如 accelerometer noise density 是 60 µg/√Hz，UrbanNav输出是400 Hz, 这个standard deviation 就是 60*√400=600 µg = 0.6mg，跟你这个里面参数的量级差不多。
具体一些单位换算可以参考这个[链接](https://www.vectornav.com/resources/inertial-navigation-primer/specifications--and--error-budgets/specs-imuspecs?gad_source=1&gclid=CjwKCAjwydSzBhBOEiwAj0XN4CMqoyJWaomPA04ABOfyO0fmMPBA8ywC2b5LUY47j7v7QKj7Bjve_RoClIMQAvD_BwE)， 

> [#59 (comment)](https://github.com/IPNL-POLYU/UrbanNavDataset/issues/59#issuecomment-2173281723) 老师，在这里，我通过您上面的提示，已经将parameter里面的acc以及gry的arw以及noise设置为了相应的文件中的信息，可能我没有表示清楚我的问题，实现中 还需要提供IMU设备的陀螺仪以及加速度计的bias的std标准差数据 基于代码作者的设置如下：（这个标准差数据 我不知道我提供的参数图片信息是否可以利用，还是这些std参数应该如何获取） ![2024-06-18 11-24-49 的屏幕截图](https://private-user-images.githubusercontent.com/74706451/340526843-897ed1a7-bd28-4374-a37b-f69447142c4b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkwMzYxOTgsIm5iZiI6MTcxOTAzNTg5OCwicGF0aCI6Ii83NDcwNjQ1MS8zNDA1MjY4NDMtODk3ZWQxYTctYmQyOC00Mzc0LWEzN2ItZjY5NDQ3MTQyYzRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjIyVDA1NTgxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI5MDZkY2Q4Zjg2ODk2OGE5YzQwZTgxNzZlOGRmMjk3ZDE1ZWQxODc3OWUxNDA5MjdjMDYyOWQ2NmZlNWZmODEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.x802NDwX_zs4FfYXUYaDgCXcioLpHkkUmuP_9XDwcRI)

hello299, 2024-06-24:
好的，谢谢老师，方便再次发一下那个参考的链接么https://github.com/IPNL-POLYU/UrbanNavDataset/issues/59#issuecomment-2183838078，这个连接我打开显示Error

DarrenWong, 2024-06-24:
抱歉，这个correlation time我在Xsens的manual里面也查不到相关信息，据我所知我们没有去定义这个的， https://www.xsens.com/hubfs/Downloads/usermanual/MTi_usermanual.pdf

hello299, 2024-06-25:
好的 谢谢老师哦～

hello299, 2024-06-28:
请问老师，安装角定义的也是前右上坐标系么

DarrenWong, 2024-06-28:
你好，请问是说外参文件吗，我们是根据右手法则。应该叫做右前上？

hello299, 2024-06-30:
hi,老师，问您一个问题：我在OB_GINS上测试Harsh Urban的数据集，因为在ob_gins中使用的是北东地坐标系，我将外参中（body angle 以及 天线杆臂，以及真值的初始航向角）已经进行了相应的修改，同时，我将IIMU误差因子中的residual残差块对应的传感器std也根据IMU传感器提供的外参进行了调整，但是目前 我将IMU标定的vrw,arw,noise噪声放大100倍结果也不是特别好，出现了大的漂移（经纬度）。

我分析原因：可能是Harsh Urban数据集因为GNSS信号差，所以，经过RTKLIIB得到的WLS解存在多处相邻点间隔超过20s多，造成的经纬度之间大幅度漂移（不知道我分析的是否正确），

我想的解决方法是根据residual进行调整，我看实现中，这些residual的定义，我没有好的想法进行下一步的修改，请老师指导以西一下。

hello299, 2024-06-30:
[image]

hello299, 2024-06-30:
[image]
[image]
或者老师有什么好的建议么？我进行一下调整

DarrenWong, 2024-07-01:
> hi,老师，问您一个问题：我在OB_GINS上测试Harsh Urban的数据集，因为在ob_gins中使用的是北东地坐标系，我将外参中（body angle 以及 天线杆臂，以及真值的初始航向角）已经进行了相应的修改，同时，我将IIMU误差因子中的residual残差块对应的传感器std也根据IMU传感器提供的外参进行了调整，但是目前 [...]

你好，建议你可以先试一下我们的deep urban一开始一段比较开阔的场景或者半遮挡的场景，harsh urban一直都存在很大的遮挡，所以可能算法初始化或者计算会遇到很大挑战，deep urban数据如果测试ok，可以再评估在harsh urban的算法有效性。

[image]

hello299, 2024-07-04:
https://github.com/IPNL-POLYU/UrbanNavDataset/issues/59#issuecomment-2196621301
老师，根据您的外参文件提供的安装角度 考虑为：前右上坐标系，那么我只需要将0.14进行取负就可以获得北东地坐标系，但是我其实想了解一下，一般IMU到载体的旋转角 需要考虑坐标系的限定么？也就是这个旋转角有前右上 和右前下之分么？
body_T_SPAN: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 1, 0, 0, 0,
           0, 1, 0, 0,
           0, 0, 1, 0.14, 
           0, 0, 0, 1]

closed 2024-07-14 by DarrenWong