IPNL-POLYU/UrbanNavDataset #62: 请问如何通过真值获取初始的速度和航向信息？

hello299, 2024-06-17:
老师，这两个数据，在gt真值中，对应的velBdx以及AccBdx我可以理解成为初始的速度以及航向信息么？
_Originally posted by @hello299 in https://github.com/IPNL-POLYU/UrbanNavDataset/issues/59#issuecomment-2169212276_

您好，老师，这里的初始航向信息，row,pitch直接用数据集提供的，那么这个heading 航向信息应该怎么理解和分析呢？我用的数据格式是：
[image]
[image]
他们的真值提供的航向为274，初始航向设置为276，那老师这个harsh urban数据集应该如何将负数值转换成类似276的形式呢

DarrenWong, 2024-06-17:
真值heading是顺时针为正，正北为零。你可以参考你这个RPY的定义，可能需要要转换成逆时针的角度或者弧度。

hello299, 2024-06-17:
好的，谢谢老师，我通过将真值轨迹地图画面展示，图片如下
[image]
因为真值的第一行数据显示的是-9.+度，所以，根据老师的讲解，我认为我可以设置为：350.+度（将负变成正）
还请老师评判一下，我的思路以及我得出的结论是否正确。

DarrenWong, 2024-06-17:
轨迹看来ok啊，关于角度，要看你的软件定义来看，如果是逆时针为正就是你这个转换

hello299, 2024-06-17:
[image]
老师，这个里面，存在着GNSS天线到IMU的杆臂信息，
同时这个body_t_span我可以理解成安装角度误差么？也就是IMU到载体的旋转角？但是感觉从字面意思上看，是body载体系到真值参考系统的安装角度？那这个IMU到载体的安装角如何确定？

DarrenWong, 2024-06-17:
可以的， UrbanNav平台的IMU和真值系统之间只有高度的差别。

closed 2024-06-22 by DarrenWong