IPNL-POLYU/UrbanNavDataset #64: 关于文献中公式的一些问题：

hello299, 2024-07-15:
Hello,Teacher：
我想请教一个问题：
      文献名称为：《Factor graph optimization for GNSS/INS integration: A comparison with the extended Kalman filter 》，在该文献中，IMU测量值是在INS系给出的，同时通过从INS->LOCAL在从local->global转到了ECEF的global系统进行分析实现，GraphGNSSLib中给的local系统为东北天坐标系，那么从INS的body系统到local系统按照右手定则的旋转顺序，不应该是z,x,y所以，对应的IMU --->local系统的旋转矩阵是：
[image]
[image]

但是，按照论文扉页的部分，旋转矩阵从INS的body->local是通过z,y,x对应实现的，这里我不太明白我分析的哪里有问题
[image]

hello299, 2024-07-15:
具体参考的东北天坐标系到IMU body系统的链接[https://blog.csdn.net/mao834099514/article/details/125083320](url)

DarrenWong, 2024-07-17:
关于body frame转换到local frame，可以参考Novatel的文档。的确Z X Y和Z Y X的形式会有所不同，但是最后效果基本差不多，我们目前的数据是地面车，所以roll和pitch的变化比较小。建议可以直接用novatel文档里面的这个矩阵。

[apn037.pdf](https://github.com/user-attachments/files/16261728/apn037.pdf)

[image]

hello299, 2024-07-29:
谢谢老师，已经知晓

closed 2024-08-18 by DarrenWong