IPNL-POLYU/UrbanNavDataset #65: Questions about the FGO test in the benchmark article

MrTHMX, 2024-08-07:
Thank you for your work! I want to know whether the FGO test in the benchmark article does not use INS? I feel that the results of the Harsh dataset are really beautiful. Can you share with me how you did it?

Pirkaklo, 2024-08-08:
> Thank you for your work! I want to know whether the FGO test in the benchmark article does not use INS? I feel that the results of the Harsh dataset are really beautiful. Can you share with me how you did it?

Hi MrTHMX, thanks for your interest in our datasets! The Harsh dataset results were evaluated by the previous ground truth, at that time, many significant errors were excluded by the old version of the ground truth and evaluation matrix and you can see the availability is not close to 100%. Thus the results of FGO are excellent.

After we update the ground truth, the mean error of F9P data should be about 12 meters in the Harsh dataset. You can see the results in our latest paper "Y. Zhong, R. Hu, X. Bai, X. Li, L. -T. Hsu and W. Wen, "Enhancing GNSS Positioning Accuracy for Road Monitoring Systems: A Factor Graph Optimization Approach Aided by Geospatial Information," in _IEEE Transactions on Instrumentation and Measurement_, vol. 73, pp. 1-12, 2024, Art no. 5501812, doi: 10.1109/TIM.2024.3369156."

Please feel free to contact me if you have any problems with our datasets or the FGO algorithm.

MrTHMX, 2024-08-10:
Has it changed in other datasets?

Pirkaklo, 2024-08-10:
> Has it changed in other datasets?

Hi MrTHMX, there is no change for other datasets currently.

MrTHMX, 2024-08-10:
cost function"Enhancing GNSS Positioning Accuracy for Road Monitoring Systems"  may be not same as what is in the benchmark?

DarrenWong, 2024-08-18:
Please refer to "Enhancing GNSS Positioning Accuracy for Road Monitoring Systems".

closed 2024-08-18 by DarrenWong