IPNL-POLYU/UrbanNavDataset #67: How to download the 3D PointCloud prior map?

maSiki29, 2024-08-23:
Hello,

 I would like to download the point cloud map at the column named "3D PointCloud" in this section [https://github.com/IPNL-POLYU/UrbanNavDataset?tab=readme-ov-file#datasets](url) 
 However, I couldn't find any valid download link on the website or this github repo. The Potree website only has some export buttons (as shown in the following image), but they don’t work. Is there any available way to download the point cloud map? Thanks!

[image]

DarrenWong, 2024-08-25:
Thank you for your interest in our data. Please check the below link for the point cloud map for the UrbanNav dataset. You can downsample the point cloud if needed using the script inside the folder if needed.

https://www.dropbox.com/scl/fo/bpawd1lh4q6dnst70nz0x/h?rlkey=mlol5haclievtn7bj2y1r2xgc&dl=0

maSiki29, 2024-08-25:
Thanks for the reply and your great work!

DarrenWong, 2024-08-26:
Can I know the meaning of point cloud frame? The map data is saved as PCD files. You can read it and publish as any frame_id using ROS accourding Section4 in  http://wiki.ros.org/pcl_ros
[image]

maSiki29, 2024-08-26:
I use the "point cloud frame" to describe the point cloud measurement by the lidar at the timestamp t. From my understanding, a stream of point cloud frame is saved in the ```/velodyne_points```. 
Each PointCloud2 message has frame_id to indicate the coordinate system at the timestamp. And my previous question is that if the point cloud map is obtained by transforming these point cloud frames to the map coordinate.

maSiki29, 2024-08-26:
I explain why I want to check the map generation. I would like to use this dataset in a particle filter-based localization work.
In this work, the particle weight is updated by computing the similarity between the real-time point cloud frame at time t and the particle measurement under the map. 
Most of works obtain the high-definition (HD) prior map in advance, then collect the real-time point cloud and gt pose on the same road again. Finally, the work match the real-time measurement and map to do localization.

So I think it doesn't make sense if the map and real-time point cloud frame (in the ```/velodyne_points```) come from the same source.

DarrenWong, 2024-08-26:
> I use the "point cloud frame" to describe the point cloud measurement by the lidar at the timestamp t. From my understanding, a stream of point cloud frame is saved in the `/velodyne_points`. Each [...]

Yes, the map is accumulated by the point cloud and the corresponding pose

DarrenWong, 2024-08-26:
> I explain why I want to check the map generation. I would like to use this dataset in a particle filter-based localization work. In this work, the particle weight is updated by computing the similarity between the real-time point cloud frame at time t and the particle measurement under the map. Most of works obtain the high-definition (HD) prior map in advance, then collect the real-time point cloud and gt pose on the same road again. Finally, the work match the real-time measurement and HD map to do localization.
> 
> So I think it doesn't make sense if the map and real-time point cloud frame (in the `/velodyne_points`) come from the same source.

Thanks for your interest in our data. We do not provide a prior map in existing open-source data and the map provided is for visualization only. You can refer to some datasets that include a prior map such as new college which includes a prior map collected by survey grade BLK360 LiDAR scanner [https://ori-drs.github.io/newer-college-dataset/]

maSiki29, 2024-08-26:
Thank you for the information!

closed 2024-08-26 by maSiki29