IPNL-POLYU/UrbanNavDataset #68: about GT

nothing371442, 2024-09-11:
Hello, thank you very much for your work!
May I ask why Novatel SPAN-CPT can still output centimeter level accuracy in high difficulty scenarios such as urban canyons? Have you done any post-processing work on Novatel SPAN-CPT output to optimize the ground truth?
In my understanding, Novatel SPAN-CPT has high-precision inertial navigation and high-precision RTK, but high-precision RTK is not suitable for challenging scenarios. Can high-precision inertial navigation (another set of sensors besides IMU sensors) be used alone to obtain high-precision, long-term, and error free positioning results?
thank you!

DarrenWong, 2024-09-13:
Thank you for your interest in our dataset. We collect GNSS and IMU raw measurements from the SPAN-CPT and postprocess the data using state-of-the-art Inertial Explorer software from NovAtel, which maximizes the accuracy of the trajectory by processing forward and backward in time, performing a backward smoothing step, and combining the results. 

Regarding your question "Can high-precision inertial navigation (another set of sensors besides IMU sensors) be used alone to obtain high-precision, long-term, and error free positioning results?"

Ans: We observed that our existing SPAN-CPT might not be good enough to provide long-term drift-free performance in super-dense urban or tunnel environments. However, it should be good enough to provide the GT in existing datasets. We also working on combining the measurements of SPAN-CPT and LiDAR/camera to achieve long-term and drift-free positioning results in urban areas. Please stay tuned!

For more details related to GT, you can refer to Section 3.1 of the [Ubrannav paper](https://navi.ion.org/content/70/4/navi.602)

nothing371442, 2024-09-13:
Very clear, thank you very much!

closed 2024-09-13 by nothing371442