IPNL-POLYU/UrbanNavDataset #69: 数据集下载的问题

lvpengxiang, 2024-09-13:
老师您好，非常感谢你们团队开源这么优秀全面的数据集，我尝试下载这个序列的rosbag，https://github.com/IPNL-POLYU/UrbanNavDataset#urbannav-hk-deep-urban-1，大小是64G，我尝试了好几次都是下载大约40个G的时候就自动结束了，请问下老师有什么解决办法吗

DarrenWong, 2024-09-15:
你好 请问你是用什么浏览器和哪个链接下来的，因为bag比较大有时候网络可能会中断导致需要重新下载，如果你是用dropbox client或者百度网盘可以支持断点续传应该会好很多。谢谢你的反馈 我们也在考虑把Dropbox链接的包切成多个20GB的压缩包，下周会更新上来。

此外，你可以查看readme里面的百度网盘链接，
For Deep Urban Whampoa,
[UrbanNav-HK-Deep-Urban-1](https://pan.baidu.com/s/1MZh2VsrValTMqzLweRs8pg)  https://pan.baidu.com/s/1MZh2VsrValTMqzLweRs8pg (ROS) (jnsa)
For Harsh Urban Mongkok
[UrbanNav-HK-Harsh-Urban-1](https://pan.baidu.com/s/16pHxF4XcbhMUXMDd2xggyQ) https://pan.baidu.com/s/16pHxF4XcbhMUXMDd2xggyQ(ROS) (2q4j)

[image]

lvpengxiang, 2024-09-15:
老师，我用的是microsoft edge浏览器下载的，链接是https://www.dropbox.com/s/1g3dllvdrgihkij/UrbanNav-HK_Whampoa-20210521_sensors.bag?dl=0

请问一下这个百度网盘链接下载下来后可以转换成rosbag吗

DarrenWong, 2024-09-15:
百度网盘跟dropbox都是一样的rosbag，百度网盘可能切成几部分，到时候点开就能解压了

lvpengxiang, 2024-09-15:
好的，感谢老师耐心解答，我现在试一下。

lvpengxiang, 2024-09-18:
非常感谢老师，已成功下载

DarrenWong, 2024-09-20:
Thanks for your feedback, we updated the Dropbox link to support the split files download, please feel free to evaluate. https://github.com/IPNL-POLYU/UrbanNavDataset/commit/d6caebe97076d6abc6eea0bdebc8492b7799ca53

closed 2024-09-20 by DarrenWong