IPNL-POLYU/UrbanNavDataset #80: Regarding the antenna used by Ublox

Erika1kuta, 2025-05-02:
作者您好，非常感谢你们的工作，对我们实验带来非常大的帮助。

我想请问你们U-blox获取的观测文件是用的 Ublox 自带的天线获得的吗？因为我看 ECMD 的数据集中（Dense Street Circle）同样是 F9P，也是在 Deep Whampoa 数据集区域采集， 但不知道为什么，卫星数量好像比 Urbannav 少很多（Urbannav 大概是30多颗，ECMD 只有15颗左右）

期待您的回复！再次感谢你们的工作

DarrenWong, 2025-05-05:
你好 请问可以方便提供分别是UrbanNav和ECMD哪两个文件吗？ Whampoa场景比较多样，UrbanNav中Whampoa场景中，Deep Urban一开始是从一个比较开阔的场景进行，然后再进去都市，星就会很少 会存在比15颗更少的情况。

Erika1kuta, 2025-05-05:
老师您好，感谢您的回复，Urbannav 中为：UrbanNav-HK-Deep-Urban-1.ublox.f9p; ECMD 中为 F9P_GNSS 文件夹下的 Dense_street_day_difficult_circle.obs
谢谢

DarrenWong, 2025-05-12:
你好，这个场景我们对比过，确实发现 UrbanNav 的星数会多一些。这是因为它们是在不同时间采集的数据，使用的天线也是 F9P 自带的多频天线。另外，也有可能是因为使用的 F9P 版本不同，我们实验室的 F9P 采购批次较多。感谢您发现这个情况，我们会在后续的不同时间采集中进一步确认这个情况。

[image]

[image]

closed 2025-05-12 by Erika1kuta