IPNL-POLYU/UrbanNavDataset #82: 真值处理的代码有bug

zhao-zhibo, 2025-08-12:
https://github.com/IPNL-POLYU/UrbanNavDataset/blob/73596f3fbe45b24bfd22b9e5b952e27d2ec84c43/tools/gt_vis/convert_novatel_to_pose.py#L37

当convert_body_frame为`true`时，求T_body_2_ENU 变量，需要将SPANIMU在n系下的位姿转换到body(LiDAR)系在n系下的位姿，公式应该是
`  T_body_2_ENU = T_ENU_2_SPAN_IMU * T_body_to_SPANIMU`。

[image]

也就是说之前的代码中将0   0   0.42，直接赋值给变量T_body_2_ENU 是不合理的，从上面的图片可知直接赋值是错误的，因此我对这部分代码做了修改,  如下:
[image]
同理pose的计算也有类似的问题,修改如下:
[image]

Commit提交修改:https://github.com/IPNL-POLYU/UrbanNavDataset/commit/5c35ae786bac465818fa97be8fda37bd23866d57

DarrenWong, 2025-08-13:
你说得很对，直接赋值 高度+ 0.42 确实是不合理的。刚好因为这是车体基本垂直于地面，所以之前计算可能差的不多。

你这边方便提一个pull request 吗？我把你的comments合并到我们主分支上。非常感谢你的认真和细心！

zhao-zhibo, 2025-08-15:
两处不合理的部分已经修改，修改如下:https://github.com/IPNL-POLYU/UrbanNavDataset/pull/83  ，现将issue关闭

closed 2025-08-15 by zhao-zhibo